wiiznokes/fan-control #91: Add AIO pump/RGB control

GrassButler, 2024-03-29:
I’ve been waiting for something like this since GWE stopped being maintained, I was hoping system76 would implement something like this so this is awesome to see. 

Would be the most perfect app ever if you could integrate AIO pump control and maybe even the led/rgb control using liquidctl? Just a suggestion!

looks amazing tho. Can’t wait to use it

wiiznokes, 2024-03-29:
Thanks! 

Support for water cooling would be handsome, but unlikely to happen before a long time, because i don't have one for now. Anyone could contribute that though!
Also, it don't seems to have any lib/crate that do this, but i probably miss-checked (using python cli/lib is not an option, except maybe if it can send data through network, but that wouldn't be a nice solution).

> maybe even the led/rgb control using liquidctl?

Support for led seems a little out of scope, it will require some change in the current architecture, and you will probably always have better customization/support with openRGB, but i understand why this is wanted. Tho, i think openRGB have an API to connect to it, maybe we could use that, but i can't say i will work on this.

DGMurdockIII, 2025-04-29:
I won't this as well