wiiznokes/fan-control #98: Linux Mint 21.3 - Not working

Hoerli1337, 2024-04-28:
Hi!
I have installed the application according to the instructions.
1. `sudo apt install lm-sensors`

2. ```
wget https://raw.githubusercontent.com/wiiznokes/fan-control/master/resource/linux/60-fan-control.rules
sudo mv 60-fan-control.rules /usr/lib/udev/rules.d
sudo udevadm control --reload-rules && sudo udevadm trigger

```
3. Install over Application management
4. Start fan-control >> Nothing :(

Running all in a terminal:
```
/usr/bin/flatpak run --branch=stable --arch=x86_64 --command=fan-control io.github.wiiznokes.fan-control
thread 'main' panicked at /run/build/fan-control/cargo/vendor/iced_tiny_skia/src/window/compositor.rs:56:10:
Create softbuffer surface for window: PlatformError(Some("Visual 0xb3 does not use softbuffer's pixel format and is unsupported"), None)
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
[ERROR hardware::linux] can't set auto to a pwm sensor when quitting: sensors_set_value() failed: [4] Kernel interface error
[ERROR hardware::linux] can't set auto to a pwm sensor when quitting: sensors_set_value() failed: [4] Kernel interface error
[ERROR hardware::linux] can't set auto to a pwm sensor when quitting: sensors_set_value() failed: [4] Kernel interface error
[ERROR hardware::linux] can't set auto to a pwm sensor when quitting: sensors_set_value() failed: [4] Kernel interface error
[ERROR hardware::linux] can't set auto to a pwm sensor when quitting: sensors_set_value() failed: [4] Kernel interface error
[ERROR hardware::linux] can't set auto to a pwm sensor when quitting: sensors_set_value() failed: [4] Kernel interface error
[ERROR hardware::linux] can't set auto to a pwm sensor when quitting: sensors_set_value() failed: [4] Kernel interface error

```

System:
CPU: AMD Ryzen R9 3900X
RAM: 32GB DDR4
GPU: AMD Radeon RX 5700XT
Board: ASUS Strix X570

wiiznokes, 2024-04-28:
Hi,
the panic indicate that the renderer doesn't support your display apparently. I don't really know much in this but maybe it could be fixed with wayland, a fix in softbuffer, or switching to wgpu. But wgpu is currently deactivate because it cause seg fault on flatpak.

The other errors probably indicate that the udev rules doesn't works.

if you really want to use fan-control now, you can try this:

If you really want to use fan-control now, you can try this (with no UI):

```
# change 7 with the right number (depend on the PC)
sudo chmod -R a+x /sys/class/hwmon/hwmon7
flatpak run io.github.wiiznokes.fan-control --cli
```

Hoerli1337, 2024-04-29:
Thanks for the info.
At the moment I can still make do with LibreFan.
The project was just discontinued, so I wanted to give fan-control a try.

Hopefully you will get it fixed :)

darkraisisi, 2024-05-01:
I ran into the exact same problem (minus the sensor errors). Following the same installation steps as noted by the first user and resulting in the same error.

```
/usr/bin/flatpak run --branch=stable --arch=x86_64 --command=fan-control io.github.wiiznokes.fan-control
thread 'main' panicked at /run/build/fan-control/cargo/vendor/iced_tiny_skia/src/window/compositor.rs:56:10:
Create softbuffer surface for window: PlatformError(Some("Visual 0x23 does not use softbuffer's pixel format and is unsupported"), None)
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```
OS:
Distro: Pop-os 22.04 LTS
Windowing system: X11

System:
CPU: AMD Ryzen 7 2700X
GPU: Nvidia RTX 3090
Board: Asus prime 470x pro

Display:
1440p ultrawide 21:9
No fractional scaling
HiDPI enabled

wiiznokes, 2024-05-01:
This is very unfortunate, there are already 3 users that reported this issue. I made an issue here FYI: https://github.com/rust-windowing/softbuffer/issues/215

wiiznokes, 2024-05-07:
@yggdrasil75, @Hoerli1337, @darkraisisi , @miguelccoelho 

Hi, sorry for the notif, a question was asked in the related issue: https://github.com/rust-windowing/softbuffer/issues/215

> @wiiznokes Can you please share the output of xdpyinfo on the affected system?

Please subscribe to it in the future if you want as i can't provide information

notgull, 2024-05-08:
This is somewhat strange, as I don't see `0xb3` in this list. It looks like somehow the visual `0xb3` is being passed into `softbuffer`... which it rejects because it doesn't exist.

Can you run `xtrace` with the failing application and post the logs?

darkraisisi, 2024-05-08:
I tried running `xtrace` but i honestly don't know if this error is related or not. Please inform me on how to do it properly if need be rtfm didn't help me.

```bash
xtrace /usr/bin/flatpak run --branch=stable --arch=x86_64 --command=fan-control io.github.wiiznokes.fan-control

No display name to create specified, trying :9
Error parsing xauth list data: less than three things in a line!
```

Output of `xdpyinfo`
```bash
name of display:    :1
version number:    11.0
vendor string:    The X.Org Foundation
vendor release number:    12101004
X.Org version: 1.21.1.4
maximum request size:  16777212 bytes
motion buffer size:  256
bitmap unit, bit order, padding:    32, LSBFirst, 32
image byte order:    LSBFirst
number of supported pixmap formats:    7
supported pixmap formats:
    depth 1, bits_per_pixel 1, scanline_pad 32
    depth 4, bits_per_pixel 8, scanline_pad 32
    depth 8, bits_per_pixel 8, scanline_pad 32
    depth 15, bits_per_pixel 16, scanline_pad 32
    depth 16, bits_per_pixel 16, scanline_pad 32
    depth 24, bits_per_pixel 32, scanline_pad 32
    depth 32, bits_per_pixel 32, scanline_pad 32
keycode range:    minimum 8, maximum 255
focus:  window 0x60048d7, revert to Parent
number of extensions:    31
    BIG-REQUESTS
    Composite
    DAMAGE
    DOUBLE-BUFFER
    DPMS
    DRI2
    DRI3
    GLX
    Generic Event Extension
    MIT-SCREEN-SAVER
    MIT-SHM
    NV-CONTROL
    NV-GLX
    Present
    RANDR
    RECORD
    RENDER
    SECURITY
    SHAPE
    SYNC
    X-Resource
    XC-MISC
    XFIXES
    XFree86-DGA
    XFree86-VidModeExtension
    XINERAMA
    XINERAMA
    XInputExtension
    XKEYBOARD
    XTEST
    XVideo
default screen number:    0
number of screens:    1

screen #0:
  dimensions:    3440x1440 pixels (910x381 millimeters)
  resolution:    96x96 dots per inch
  depths (7):    24, 1, 4, 8, 15, 16, 32
[... 914 lines cut ...]
    available colormap entries:    256 per subfield
    red, green, blue masks:    0xff0000, 0xff00, 0xff
    significant bits in color specification:    8 bits
  visual:
    visual id:    0xa2
    class:    TrueColor
    depth:    32 planes
    available colormap entries:    256 per subfield
    red, green, blue masks:    0xff0000, 0xff00, 0xff
    significant bits in color specification:    8 bits
  visual:
    visual id:    0xa3
    class:    TrueColor
    depth:    32 planes
    available colormap entries:    256 per subfield
    red, green, blue masks:    0xff0000, 0xff00, 0xff
    significant bits in color specification:    8 bits
  visual:
    visual id:    0xa4
    class:    TrueColor
    depth:    32 planes
    available colormap entries:    256 per subfield
    red, green, blue masks:    0xff0000, 0xff00, 0xff
    significant bits in color specification:    8 bits
```

Xcaliburr101, 2024-05-10:
i have the same issue
`thread 'main' panicked at /run/build/fan-control/cargo/vendor/iced_tiny_skia/src/window/compositor.rs:56:10:
Create softbuffer surface for window: PlatformError(Some("Visual 0x23 does not use softbuffer's pixel format and is unsupported"), None)
stack backtrace:
   0:           0xde3ba6 - <unknown>
   1:           0xe12460 - <unknown>
   2:           0xddfd8f - <unknown>
   3:           0xde3984 - <unknown>
   4:           0xde5207 - <unknown>
   5:           0xde4f69 - <unknown>
   6:           0xde5698 - <unknown>
   7:           0xde5572 - <unknown>
   8:           0xde40a6 - <unknown>
   9:           0xde52c4 - <unknown>
  10:           0x44b455 - <unknown>
  11:           0x44b9a3 - <unknown>
  12:           0x4c99f8 - <unknown>
  13:           0x4a7806 - <unknown>
  14:           0x4a8a3d - <unknown>
  15:           0x4621e9 - <unknown>
  16:           0x52ebe3 - <unknown>
  17:           0x51dee9 - <unknown>
  18:           0xdd5e31 - <unknown>
  19:           0x462b25 - <unknown>
  20:     0x7335c483108a - <unknown>
  21:     0x7335c483114b - __libc_start_main
  22:           0x44c0b5 - <unknown>
  23:                0x0 - <unknown>`

Xcaliburr101, 2024-05-10:
> i **have the same issue** 
`thread 'main' panicked at /run/build/fan-control/cargo/vendor/iced_tiny_skia/src/window/compositor.rs:56:10: Create softbuffer surface for window: PlatformError(Some("Visual 0x23 does not use softbuffer's pixel format and is unsupported"), None) stack backtrace: 0: 0xde3ba6 - <unknown> 1: 0xe12460 - <unknown> 2: 0xddfd8f - <unknown> 3: 0xde3984 - <unknown> 4: 0xde5207 - <unknown> 5: 0xde4f69 - <unknown> 6: 0xde5698 - <unknown> 7: 0xde5572 - <unknown> 8: 0xde40a6 - <unknown> 9: 0xde52c4 - <unknown> 10: 0x44b455 - <unknown> 11: 0x44b9a3 - <unknown> 12: 0x4c99f8 - <unknown> 13: 0x4a7806 - <unknown> 14: 0x4a8a3d - <unknown> 15: 0x4621e9 - <unknown> 16: 0x52ebe3 - <unknown> 17: 0x51dee9 - <unknown> 18: 0xdd5e31 - <unknown> 19: 0x462b25 - <unknown> 20: 0x7335c483108a - <unknown> 21: 0x7335c483114b - __libc_start_main 22: 0x44c0b5 - <unknown> 23: 0x0 - <unknown>`

**also running the cli version askes for a config file, where can i find this?**
`flatpak run io.github.wiiznokes.fan-control --cli
There is no active configuration defined
You can define one in /home/popper/.var/app/io.github.wiiznokes.fan-control/config/fan-control directory,
Or specifie a config to use using an argument`

wiiznokes, 2024-05-10:
> also running the cli version askes for a config file, where can i find this?

The are example in [this directory](https://github.com/wiiznokes/fan-control/tree/master/.config). I plan to add some doc for it some day. Don't forget to change `current_config` in `settings.toml`, or point to it via command line (see the  `--help` option).

Please, open another issue if you have further questions

wiiznokes, 2024-05-10:
@notgull I also get this bug with x11

Vaaarna, 2024-05-13:
same issue, i will blow my brains bc fan is right by my face and has no chill (statement for comedic effect and to add a +1 to the issue, i am chilling and wish yall a good day)

wiiznokes, 2024-05-17:
hi all, i've just pushed an update on flathub, with the wgpu feature activated. This should solve this issue for most of you.

zhangjing-GitHub-Code, 2024-05-19:
Well, On my Arch Linux x86_64, when I login KDE6 with X11, it has this error, but the app starts when I use Wayland.
Wiiznokes, are you using x11 or wayland?

wiiznokes, 2024-05-19:
The bug is only present on x11. But now that wgpu is active, if you have a GPU, you should not be affected