wiiznokes/fan-control #102: Build From Source install  error

1412kauti, 2024-04-30:
I was using the build instructions provided

```
git clone https://github.com/wiiznokes/fan-control.git
cd fan-control
sudo apt install make bison flex clang -y
just libsensors
cargo run --release
```
Things were perfect till the last step
`cargo run --release`

The error I got is
```
❯ cargo run --release
   Compiling sensors-sys v0.2.12
   Compiling fluent-langneg v0.13.0
   Compiling intl_pluralrules v7.0.2
   Compiling ouroboros_macro v0.17.2
   Compiling dirs v5.0.1
   Compiling darling_core v0.20.8
   Compiling iced_style v0.12.0 (https://github.com/pop-os/libcosmic?branch=master#61a14a95)
   Compiling self_cell v0.10.3
   Compiling wayland-protocols-plasma v0.2.0
   Compiling tempfile v3.10.1
   Compiling xkbcommon-dl v0.4.2
   Compiling fluent-syntax v0.11.0
   Compiling crossbeam-channel v0.5.12
   Compiling filetime v0.2.23
   Compiling xdg v2.5.2
   Compiling equivalent v1.0.1
   Compiling utf8parse v0.2.1
   Compiling unicode-xid v0.2.4
   Compiling cargo-packager-resource-resolver v0.1.0
   Compiling derive_more-impl v1.0.0-beta.6
   Compiling notify v6.1.1
   Compiling anstyle-parse v0.2.3
   Compiling indexmap v2.2.6
   Compiling fluent-bundle v0.15.2
   Compiling atomicwrites v0.4.2 (https://github.com/jackpot51/rust-atomicwrites#043ab485)
   Compiling ordered-multimap v0.7.1
   Compiling num-rational v0.4.1
   Compiling cached_proc_macro v0.20.0
   Compiling fluent v0.16.0
   Compiling num_enum v0.7.2
   Compiling iced_renderer v0.12.0 (https://github.com/pop-os/libcosmic?branch=master#61a14a95)
   Compiling parking_lot v0.12.1
   Compiling rust-embed-utils v8.3.0
   Compiling ouroboros_macro v0.18.3
   Compiling num-iter v0.1.44
   Compiling ouroboros v0.17.2
   Compiling winit v0.29.10 (https://github.com/pop-os/winit.git?branch=winit-0.29#bdc66109)
   Compiling rust-embed v8.3.0
   Compiling ron v0.8.1
   Compiling cargo-packager-utils v0.1.0
   Compiling unicode-normalization v0.1.23
   Compiling cosmic-config-derive v0.1.0 (https://github.com/pop-os/libcosmic?branch=master#61a14a95)
   Compiling serde_spanned v0.6.5
   Compiling toml_datetime v0.6.5
   Compiling num-complex v0.4.5
   Compiling instant v0.1.12
   Compiling cached_proc_macro_types v0.1.1
   Compiling arc-swap v1.7.1
   Compiling serde_json v1.0.115
   Compiling anstyle-query v1.0.2
   Compiling darling_macro v0.20.8
   Compiling signal-hook v0.3.17
   Compiling winnow v0.6.5
   Compiling anstyle v1.0.6
   Compiling colorchoice v1.0.0
[... 107 lines cut ...]
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/panicking.rs:516:19
  26:           0xdcd507 - std::panic::catch_unwind::h24c28c23c02c3841
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/panic.rs:142:14
  27:           0xdcd507 - std::rt::lang_start_internal::{{closure}}::h705d3c9cbc06ef47
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/rt.rs:148:48
  28:           0xdcd507 - std::panicking::try::do_call::ha21f52ba13158470
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/panicking.rs:552:40
  29:           0xdcd507 - std::panicking::try::h5581346bf6aeb1f8
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/panicking.rs:516:19
  30:           0xdcd507 - std::panic::catch_unwind::h7919645a6b72e25b
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/panic.rs:142:14
  31:           0xdcd507 - std::rt::lang_start_internal::h12de51168669836e
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/rt.rs:148:20
  32:           0x5676ce - std::rt::lang_start::h236675407cae54ea
  33:     0x7b74a3229d90 - <unknown>
  34:     0x7b74a3229e40 - __libc_start_main
  35:           0x44d435 - _start
  36:                0x0 - <unknown>
[ERROR hardware::linux] can't set auto to a pwm sensor when quitting: sensors_set_value() failed: [4] Kernel interface error
[ERROR hardware::linux] can't set auto to a pwm sensor when quitting: sensors_set_value() failed: [4] Kernel interface error
[ERROR hardware::linux] can't set auto to a pwm sensor when quitting: sensors_set_value() failed: [4] Kernel interface error
[ERROR hardware::linux] can't set auto to a pwm sensor when quitting: sensors_set_value() failed: [4] Kernel interface error

```
I'm not familiar with Rust...a little help plz

wiiznokes, 2024-04-30:
This is a runtime error. The same of #98 .

I don't think it can be fixed but i have some question that might help

- are you running wayland ?
- what DE do you use and which version / Linux distrib (`neofetch`)

You could try to modify `ui/Cargo.toml`, and uncomment `#"wgpu",`, ei, change to `"wgpu",` and rerun `cargo run`.

Also, have you installed the udev rules ?

1412kauti, 2024-05-03:
Hey!...
Im using PopOS22.04 Default DE...not the new Cosmic DE 
No wayland....

Editting the Cargo.toml worked...thankx

closed 2024-05-03 by 1412kauti