wiiznokes/fan-control #139: PID control

Xavantex, 2024-06-21:
I just want to preface this issue that I love that this project, from what I gather after some superficial investigation, does not actually need the lm-sensors package with the necessary libraries embedded into the binary, and that its rust so its a generally small footprint.

#### Use Case
PID control would adapt fan speed to a desired temperature point.
This would stop temperature spikes to create fan spikes, and make fans try to reach a desired temperature if possible.
It would also spin down fans if temperature is already at desired temperature.

#### Proposed Change
New PID behavior for control.

#### Who Benefits From The Change(s) ?
Those who wish to maintain constant temperature or avoid fans spinning up and down too quickly.

wiiznokes, 2024-06-21:
> I just want to preface this issue that I love that this project, from what I gather after some superficial investigation, does not actually need the lm-sensors package with the necessary libraries embedded into the binary, and that its rust so its a generally small footprint.

Thanks, yeah this is the case. We couldn't rely on libsensors system dep because we use a custom fork anyway.
> PID control would adapt fan speed to a desired temperature point.
This would stop temperature spikes to create fan spikes, and make fans try to reach a desired temperature if possible.
It would also spin down fans if temperature is already at desired temperature.

Is there an equivalent in this project https://getfancontrol.com/ ?

It sound like this could be done using the Graph behavior, maybe with a settings for not make a maximum delta for changing percentage, e.i, to go from 20 to 50%, it must make multiple step each iteration (25, 30, ...)

Xavantex, 2024-06-21:
As referenced here: https://github.com/Rem0o/FanControl.Releases/issues/2196#issuecomment-1857191868
and here: https://github.com/Rem0o/FanControl.Releases/issues/2045#issuecomment-1735678045

I think it's safe to say the repo owner do not believe in the concept, or believe it possible.
There have been some implementations, for example https://github.com/markusressel/fan2go and https://github.com/mrstux/hybrid_fan_control / https://github.com/khorton/nas_fan_control?tab=readme-ov-file

I think while @Rem0o raises a few valid points, it should be a matter of careful configuration and not something someone without experience with PIDs should do, or at the very least very much at their own risk.

wiiznokes, 2024-06-22:
Ok so i did miss understood the first comment. I don't have strong opinion on adding a voltage sensor. It seems to have request for it so it might be a good addition.

I'm not sure how this would integrate in fan-control tho? 
Can we just provide a voltage sensor similar to the temp sensor and that it ? We could plug it on, let's say the graph behavior. This will be relatively easy to implement i think.
Or for the voltage sensor to be useful, we also need to map it to a temp sensor, and do more complicated stuff? This will increase the complexity of fan-control more.

So it will help if you could explain what exactly fan-control need to do.