wiiznokes/fan-control #148: Blank screen on Ubuntu 24

chulanovskyi, 2024-07-17:
#### Bug Description
* simply running the app after a fresh install - doesn't work

#### Expected Result
* the app works

#### Steps to Reproduce
* install the app according to the instruction from a readme

#### Operating system
```
Linux oleg-pc 6.8.0-38-generic #38-Ubuntu SMP PREEMPT_DYNAMIC Fri Jun  7 15:25:01 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux

Intel(R) Core(TM) i7-14700K
GeForce GTX 1660 Ti Driver Version: 535.183.01
```

#### fan-control version
* latest

#### Logs
```
oleg@oleg-pc:~$ RUST_BACKTRACE=full flatpak run io.github.wiiznokes.fan-control 
[ERROR winit::platform_impl::platform] X11 error: XError {
        description: "BadMatch (invalid parameter attributes)",
        error_code: 8,
        request_code: 149,
        minor_code: 4,
    }
[ERROR winit::platform_impl::platform] X11 error: XError {
        description: "BadMatch (invalid parameter attributes)",
        error_code: 8,
        request_code: 149,
        minor_code: 4,
    }
[ERROR winit::platform_impl::platform] X11 error: XError {
        description: "BadMatch (invalid parameter attributes)",
        error_code: 8,
        request_code: 149,
        minor_code: 4,
    }
[ERROR winit::platform_impl::platform] X11 error: XError {
        description: "BadMatch (invalid parameter attributes)",
        error_code: 8,
        request_code: 149,
        minor_code: 4,
    }
[ERROR winit::platform_impl::platform] X11 error: XError {
        description: "BadMatch (invalid parameter attributes)",
        error_code: 8,
        request_code: 149,
        minor_code: 4,
    }
[ERROR winit::platform_impl::platform] X11 error: XError {
        description: "BadMatch (invalid parameter attributes)",
        error_code: 8,
        request_code: 149,
        minor_code: 4,
    }
[ERROR iced_winit::multi_window] Error Timeout when presenting surface.
thread 'main' panicked at /run/build/fan-control/cargo/vendor/winit/src/platform_impl/linux/x11/event_processor.rs:1042:30:
Failed to unfocus input context: XError { description: "BadMatch (invalid parameter attributes)", error_code: 8, request_code: 149, minor_code: 4 }
stack backtrace:
   0:          0x14ea6f5 - <unknown>
   1:          0x15187eb - <unknown>
   2:          0x14e64ef - <unknown>
   3:          0x14ea4ce - <unknown>
   4:          0x14eb939 - <unknown>
   5:          0x14eb67d - <unknown>
   6:          0x14ebdd3 - <unknown>
   7:          0x14ebcb4 - <unknown>
   8:          0x14eabb9 - <unknown>
   9:          0x14eb9e7 - <unknown>
  10:           0x47a8b3 - <unknown>
  11:           0x47ad66 - <unknown>
  12:           0x4862f6 - <unknown>
  13:           0x48c126 - <unknown>
  14:           0x5e13de - <unknown>
  15:           0x56dcc0 - <unknown>
  16:           0x49b545 - <unknown>
  17:           0x49d3a3 - <unknown>
  18:           0x55fc19 - <unknown>
  19:          0x14dc9f0 - <unknown>
  20:           0x49b7d5 - <unknown>
  21:     0x731192a3e08a - <unknown>
  22:     0x731192a3e14b - __libc_start_main
  23:           0x47b3d5 - <unknown>
  24:                0x0 - <unknown>
^C

```

#### Possible Additional Information (optional)
* maybe unrelated, but I do have a `psensor` running ok, showing the temps and the GPU fans rpm

wiiznokes, 2024-07-17:
This sound like a x11 specific bug. Maybe try with wayland

chulanovskyi, 2024-07-18:
With the wayland everything is even worse, I can't even open the system settings or a nautilus.

``` 
warning: queue 0x2df27b0 destroyed while proxies still attached:
  wl_registry@2 still attached
libEGL warning: egl: failed to create dri2 screen
warning: queue 0x3923700 destroyed while proxies still attached:
  wl_registry@39 still attached
thread 'main' panicked at /run/build/fan-control/cargo/vendor/wgpu/src/backend/wgpu_core.rs:711:18:
Error in Surface::configure: Validation Error

Caused by:
    Parent device is lost

stack backtrace:
   0:          0x14ea6f5 - <unknown>
   1:          0x15187eb - <unknown>
   2:          0x14e64ef - <unknown>
   3:          0x14ea4ce - <unknown>
   4:          0x14eb939 - <unknown>
   5:          0x14eb67d - <unknown>
   6:          0x14ebdd3 - <unknown>
   7:          0x14ebcb4 - <unknown>
   8:          0x14eabb9 - <unknown>
   9:          0x14eb9e7 - <unknown>
  10:           0x47a8b3 - <unknown>
  11:           0xed4019 - <unknown>
  12:           0xed7795 - <unknown>
  13:           0xee5a63 - <unknown>
  14:           0xea96e1 - <unknown>
  15:           0xa3704e - <unknown>
  16:           0x50aa21 - <unknown>
  17:           0x558786 - <unknown>
  18:           0x56da78 - <unknown>
  19:           0x49b545 - <unknown>
  20:           0x49d3a3 - <unknown>
  21:           0x55fc19 - <unknown>
  22:          0x14dc9f0 - <unknown>
  23:           0x49b7d5 - <unknown>
  24:     0x7f349d46b08a - <unknown>
  25:     0x7f349d46b14b - __libc_start_main
  26:           0x47b3d5 - <unknown>
  27:                0x0 - <unknown>
```