wiiznokes/fan-control #168: No fans detected (at all)

qtqgyt, 2024-10-30:
_Note: Feel free to fill only the sections that feels relevant to you_

#### Bug Description
I don't know. I just installed the program, its getting all of my temp sensors, but not my fans.
#### Expected Result
Hopefully that it would... detect... my yk... fans...?
#### Steps to Reproduce
Just... install the app?
#### Possible Solutions (optional)

#### Possible Additional Information (optional)
I have a Core i5-6500, a Z170XP-SLI, 4 Corsair 8GB DDR4 sticks, and a GTX 1060. My fans consist of: The stock intel cooler, 1 Noctua fan in the rear, and in the front, 3 deepcool fans that came with the case.
It works fine with the "Fan Control" program using .NET on Windows. (Not this project)
#### Operating system
Ubuntu 24.04
#### fan-control version
24.8.23
#### Logs
collin@CATMANCOLLIN:~/Downloads$ flatpak run io.github.wiiznokes.fan-control

Literally nothing
On linux, run `flatpak run io.github.wiiznokes.fan-control` in a terminal.

wiiznokes, 2024-10-30:
Did you install the ... [udev rules](https://github.com/wiiznokes/fan-control/blob/master/res/linux/udev_rules.md) ?

MinaOnMars, 2024-11-04:
I don't know about OP, but I did install the udev rules, I followed the listed instructions on the main page for fan-control, and I don't see any fans in the GUI either. I ran the sensor check successfully and it found everything, but there are no fans or options within the program. I can only change labels or the item reporting temps in the list.

Relevant info:  i7-9700, RTX 2060, 16gb ram, Fedora 41 workstation, kernel 6.11.5, Wayland as the compositor. No error logs from the software are present. If any other info is needed just let me know!

wiiznokes, 2024-11-04:
What is the output of `pwmconfig` ?

qtqgyt, 2024-11-14:
Sorry I moved back to Windows just for the better fan control options.

closed 2024-11-14 by qtqgyt