wiiznokes/fan-control #172: No fans detected on Surface Pro 8

ssssamstrup, 2024-11-09:
#### Bug Description
No fans are detected

#### Expected Result
That the fan is detected

#### Steps to Reproduce
Install app
Add an item > Monitor a fan
Fan item is add with None
No options when click None dropdown to choose fan

#### Possible Solutions (optional)
Update app to detect Surface Pro 8 fans

#### Possible Additional Information (optional)
Surface Pro 8 16gb i7 (has a fan/s, it spins up when it gets hot)

#### Operating system
Windows 11

#### fan-control version
24.8.23 (latest, just downloaded and installed it)

#### Logs
(How to I provide logs on Windows 11? Then I'll do it as needed)

wiiznokes, 2024-11-09:
Did you try this software instead? https://github.com/Rem0o/FanControl.Releases

Maybe you will be more lucky (though we use the same backend so I doubt you will see more fans)

ssssamstrup, 2024-11-11:
Thank you for linking that. Same result.

[In fancontrol.releases' manual](https://github.com/Rem0o/FanControl.Releases#issues-and-hardware-compatibility) I found a link to the hardware library ([LibreHardwareMonitor](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor)) and a suggestion to post »missing fan issues« to that project. [I've added my issue there](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/1521), so it hopefully will get added and will start working with both fan control programs.

I'm guessing it could be an idea for you to add a similar section to this project, so people will post missing fan/hardware issues, where people are looking into fixing them, rather on this project itself. Hope you'll consider this.

wiiznokes, 2024-11-11:
Yeah, that could be a great addition

ssssamstrup, 2024-11-12:
Do you want me to close this or let it stay open (since this is not something you directly can solve)? I can also make a new one that directs people with no fans detected to submit a issue at LibreHardwareMonitor? Let me know what would be the best actions to take. :)

wiiznokes, 2024-11-12:
I think we can close this one and create a new one for redirect to lhm. I will add it to the read me when I can. Thanks

closed 2024-11-23 by wiiznokes