wiiznokes/fan-control #174: App Fails to Launch on Debian 12 (bookworm)

jackakers13, 2024-11-16:
#### Bug Description
App does not launch.

#### Expected Result
App should launch.

#### Steps to Reproduce
Try to launch the app, either through clicking on its icon or running `flatpak run io.github.wiiznokes.fan-control`

#### Additional Information
Other reviews on the GNOME Software app indicate similar issues. You can see some of them scraped here: https://paroj.github.io/odrs-browser/#io.github.wiiznokes.fan-control

#### Operating system
Debian GNU/Linux 12 (bookworm)

#### fan-control version
24.8.23 from Flathub

#### Logs
```
$ RUST_BACKTRACE=full flatpak run  io.github.wiiznokes.fan-control
thread 'main' panicked at /run/build/fan-control/cargo/vendor/wgpu/src/backend/wgpu_core.rs:711:18:
Error in Surface::configure: Validation Error

Caused by:
    Parent device is lost

stack backtrace:
   0:           0xdb799a - <unknown>
   1:           0x5bdecb - <unknown>
   2:           0xd80742 - <unknown>
   3:           0xdbc497 - <unknown>
   4:           0xdbd735 - <unknown>
   5:           0xdbcf35 - <unknown>
   6:           0xdbce99 - <unknown>
   7:           0xdbce84 - <unknown>
   8:           0x4098a2 - <unknown>
   9:          0x109e236 - <unknown>
  10:          0x10d3ea7 - <unknown>
  11:           0x9f291d - <unknown>
  12:           0x890010 - <unknown>
  13:           0x48b31e - <unknown>
  14:           0x48fbfe - <unknown>
  15:           0x89aeb0 - <unknown>
  16:           0x719df3 - <unknown>
  17:           0x892690 - <unknown>
  18:     0x7f464c03b188 - <unknown>
  19:     0x7f464c03b24b - __libc_start_main
  20:           0x46fe05 - <unknown>
  21:                0x0 - <unknown>
```

On linux, run `flatpak run io.github.wiiznokes.fan-control` in a terminal.

wiiznokes, 2024-11-16:
@jackakers13 Could you try this version ?

```
flatpak install --user https://dl.flathub.org/build-repo/144069/io.github.wiiznokes.fan-control.flatpakref
```
It contains some unreleased changes, with an updated version of wgpu.

wiiznokes, 2024-11-16:
Also, before doing that, you should try to use software rendering. You will need to set `ICED_BACKEND=tiny-skia` (you can use flatseal for that)