wiiznokes/fan-control #175: Fans speed up as soon as i close the GUI

markthwain1797, 2024-11-21:
#### Bug Description
As in title the fans speed up if I close the GUI and slow down again to the configured setting if I open the GUI again

#### Expected Result
I guess the fans should stay at the configured speed

#### Steps to Reproduce
Open and close the application

#### Additional Information

#### Operating system
Zorin OS 17.2 Core

#### fan-control version
24.8.23 from flathub

#### Logs
[ERROR hardware::linux] can't set auto to a pwm sensor when quitting: sensors_set_value() failed: [9] Can't write
[ERROR hardware::linux] can't set auto to a pwm sensor when quitting: sensors_set_value() failed: [9] Can't write
[ERROR hardware::linux] can't set auto to a pwm sensor when quitting: sensors_set_value() failed: [9] Can't write
[ERROR hardware::linux] can't set auto to a pwm sensor when quitting: sensors_set_value() failed: [9] Can't write

wiiznokes, 2024-11-23:
Hum, i saw this kind of error on my PC as well. It's weird that the app can modify the fan speed if we can't write in the first place.

Could you post the output of `ll /sys/class/hwmon/hwmon7/pwm*` ? (the number 7 can varies between motherboard)

Also, when you haven't yet use fan-control, was the fans sped up ? It's expected that when the app is closed, the fan speed go back to their initial state (i plan to implement a tray icon for running in background).

markthwain1797, 2024-11-25:
Here's my output from ll /sys/class/hwmon/hwmon3/pwm*

```
-rw-rw-rw- 1 root root 4096 Nov 25 18:21 /sys/class/hwmon/hwmon3/pwm1
-rw-rw-rw- 1 root root 4096 Nov 25 18:17 /sys/class/hwmon/hwmon3/pwm1_auto_channels_temp
-rw-rw-rw- 1 root root 4096 Nov 25 18:17 /sys/class/hwmon/hwmon3/pwm1_auto_point1_temp
-rw-rw-rw- 1 root root 4096 Nov 25 18:17 /sys/class/hwmon/hwmon3/pwm1_auto_point1_temp_hyst
-rw-rw-rw- 1 root root 4096 Nov 25 18:17 /sys/class/hwmon/hwmon3/pwm1_auto_point2_temp
-rw-rw-rw- 1 root root 4096 Nov 25 18:17 /sys/class/hwmon/hwmon3/pwm1_auto_point3_temp
-rw-rw-rw- 1 root root 4096 Nov 25 18:17 /sys/class/hwmon/hwmon3/pwm1_auto_slope
-rw-rw-rw- 1 root root 4096 Nov 25 18:17 /sys/class/hwmon/hwmon3/pwm1_auto_start
-rw-rw-rw- 1 root root 4096 Nov 25 18:21 /sys/class/hwmon/hwmon3/pwm1_enable
-rw-rw-rw- 1 root root 4096 Nov 25 18:17 /sys/class/hwmon/hwmon3/pwm1_freq
-rw-rw-rw- 1 root root 4096 Nov 25 18:21 /sys/class/hwmon/hwmon3/pwm2
-rw-rw-rw- 1 root root 4096 Nov 25 18:17 /sys/class/hwmon/hwmon3/pwm2_auto_channels_temp
-rw-rw-rw- 1 root root 4096 Nov 25 18:17 /sys/class/hwmon/hwmon3/pwm2_auto_point1_temp
-rw-rw-rw- 1 root root 4096 Nov 25 18:17 /sys/class/hwmon/hwmon3/pwm2_auto_point1_temp_hyst
-rw-rw-rw- 1 root root 4096 Nov 25 18:17 /sys/class/hwmon/hwmon3/pwm2_auto_point2_temp
-rw-rw-rw- 1 root root 4096 Nov 25 18:17 /sys/class/hwmon/hwmon3/pwm2_auto_point3_temp
-rw-rw-rw- 1 root root 4096 Nov 25 18:17 /sys/class/hwmon/hwmon3/pwm2_auto_slope
-rw-rw-rw- 1 root root 4096 Nov 25 18:17 /sys/class/hwmon/hwmon3/pwm2_auto_start
-rw-rw-rw- 1 root root 4096 Nov 25 18:21 /sys/class/hwmon/hwmon3/pwm2_enable
-rw-rw-rw- 1 root root 4096 Nov 25 18:17 /sys/class/hwmon/hwmon3/pwm2_freq
-rw-rw-rw- 1 root root 4096 Nov 25 18:21 /sys/class/hwmon/hwmon3/pwm3
-rw-rw-rw- 1 root root 4096 Nov 25 18:17 /sys/class/hwmon/hwmon3/pwm3_auto_channels_temp
-rw-rw-rw- 1 root root 4096 Nov 25 18:17 /sys/class/hwmon/hwmon3/pwm3_auto_point1_temp
-rw-rw-rw- 1 root root 4096 Nov 25 18:17 /sys/class/hwmon/hwmon3/pwm3_auto_point1_temp_hyst
-rw-rw-rw- 1 root root 4096 Nov 25 18:17 /sys/class/hwmon/hwmon3/pwm3_auto_point2_temp
-rw-rw-rw- 1 root root 4096 Nov 25 18:17 /sys/class/hwmon/hwmon3/pwm3_auto_point3_temp
-rw-rw-rw- 1 root root 4096 Nov 25 18:17 /sys/class/hwmon/hwmon3/pwm3_auto_slope
-rw-rw-rw- 1 root root 4096 Nov 25 18:17 /sys/class/hwmon/hwmon3/pwm3_auto_start
-rw-rw-rw- 1 root root 4096 Nov 25 18:21 /sys/class/hwmon/hwmon3/pwm3_enable
-rw-rw-rw- 1 root root 4096 Nov 25 18:17 /sys/class/hwmon/hwmon3/pwm3_freq
-rw-rw-rw- 1 root root 4096 Nov 25 18:21 /sys/class/hwmon/hwmon3/pwm4
-rw-rw-rw- 1 root root 4096 Nov 25 18:17 /sys/class/hwmon/hwmon3/pwm4_auto_channels_temp
-rw-rw-rw- 1 root root 4096 Nov 25 18:17 /sys/class/hwmon/hwmon3/pwm4_auto_point1_temp
-rw-rw-rw- 1 root root 4096 Nov 25 18:17 /sys/class/hwmon/hwmon3/pwm4_auto_point1_temp_hyst
-rw-rw-rw- 1 root root 4096 Nov 25 18:17 /sys/class/hwmon/hwmon3/pwm4_auto_point2_temp
-rw-rw-rw- 1 root root 4096 Nov 25 18:17 /sys/class/hwmon/hwmon3/pwm4_auto_point3_temp
-rw-rw-rw- 1 root root 4096 Nov 25 18:17 /sys/class/hwmon/hwmon3/pwm4_auto_slope
-rw-rw-rw- 1 root root 4096 Nov 25 18:21 /sys/class/hwmon/hwmon3/pwm4_enable
-rw-rw-rw- 1 root root 4096 Nov 25 18:17 /sys/class/hwmon/hwmon3/pwm4_freq
-rw-rw-rw- 1 root root 4096 Nov 25 18:17 /sys/class/hwmon/hwmon3/pwm5
-rw-rw-rw- 1 root root 4096 Nov 25 18:17 /sys/class/hwmon/hwmon3/pwm5_auto_channels_temp
-rw-rw-rw- 1 root root 4096 Nov 25 18:17 /sys/class/hwmon/hwmon3/pwm5_auto_point1_temp
-rw-rw-rw- 1 root root 4096 Nov 25 18:17 /sys/class/hwmon/hwmon3/pwm5_auto_point1_temp_hyst
-rw-rw-rw- 1 root root 4096 Nov 25 18:17 /sys/class/hwmon/hwmon3/pwm5_auto_point2_temp
-rw-rw-rw- 1 root root 4096 Nov 25 18:17 /sys/class/hwmon/hwmon3/pwm5_auto_point3_temp
-rw-rw-rw- 1 root root 4096 Nov 25 18:17 /sys/class/hwmon/hwmon3/pwm5_auto_slope
-rw-rw-rw- 1 root root 4096 Nov 25 18:21 /sys/class/hwmon/hwmon3/pwm5_enable
-rw-rw-rw- 1 root root 4096 Nov 25 18:17 /sys/class/hwmon/hwmon3/pwm5_freq
```

After boot (before opening fancontrol for the first time or if I don't open fancontrol at all) the fans use the temperature-curve configured in UEFI.

wiiznokes, 2024-11-25:
The permissions seems fine

briefer666, 2025-04-09:
same problem with linux mint

wiiznokes, 2025-04-09:
But reading the title now, it is expected that the fans go back to their previous state when closing the app, since we need the app to run to calcul the speed in real time. So just don't close the app ig

briefer666, 2025-04-09:
ok, back to the smartfan option in the bios, then

thanks