wiiznokes/fan-control #180: Emphasize "add item", expand item menu by default, or add walkthrough/pop-up

Anthonyn56, 2024-12-10:
Completely willing to admit it is a skill issue on my part.

However, it was hard to tell what I was suppose to do when the software first opened.
Additionally, as noted in another issue, the menu icon is hard to see when the app is turned to dark mode which might be the one of the first actions for somebody on first install.

wiiznokes, 2024-12-10:
Yeah, maybe we should just remove the non expanded state since it doesn't cover other items.

Also, the coloring of icons should be fixed in the next update. I'm just waiting for a bug to be fixed in libcosmic before making one

wiiznokes, 2025-07-29:
fixed by adding padding and make icons bigger. There is still an unexpanded state for a sake of not having to much buttons on screen by defaut, and avoid miss click when your config is sable

closed 2025-07-29 by wiiznokes