wiiznokes/fan-control #182: sensors detected but "Control" didn't show any of them.

yuzujr, 2025-01-31:
#### Bug Description

sensors detected but "Control" didn't show any of them.
`sensors` output:
```
❯ sensors
dell_smm-virtual-0
Adapter: Virtual device
Processor Fan:   5600 RPM  (min =    0 RPM, max = 5600 RPM)
Motherboard Fan: 5200 RPM  (min =    0 RPM, max = 5600 RPM)
CPU:              +41.0°C  
SODIMM:           +31.0°C  
Ambient:          +39.0°C  

amdgpu-pci-0600
Adapter: PCI adapter
vddgfx:        1.19 V  
vddnb:       937.00 mV 
edge:         +39.0°C  
PPT:          25.00 W  

nvme-pci-0200
Adapter: PCI adapter
Composite:    +29.9°C  (low  =  -5.2°C, high = +82.8°C)
                       (crit = +84.8°C)

iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +34.0°C  

k10temp-pci-00c3
Adapter: PCI adapter
Tctl:         +41.8°C  

nvme-pci-0500
Adapter: PCI adapter
Composite:    +27.9°C  (low  =  -0.1°C, high = +82.8°C)
                       (crit = +84.8°C)
Sensor 1:     +27.9°C  (low  = -273.1°C, high = +65261.8°C)

BAT1-acpi-0
Adapter: ACPI interface
in0:          11.97 V  
curr1:         0.00 A  
```

But,

[image]

#### Expected Result

sensors show in drop down bar

#### Steps to Reproduce

#### Additional Information

#### Operating system

Fedora 41
gnome on x11

#### fan-control version

24.8.23

#### Logs

On linux, run `flatpak run io.github.wiiznokes.fan-control` in a terminal.

yuzujr, 2025-01-31:
Although this has nothing to do with this problem, I would like to ask a question. I have a dual system of windows and linux. My fan management seems to **depend on windows**?
just now i reboot several times in linux,and the fan kept working at full speed.
this is why I'm looking for a fan control software.
After I start windows once and reboot into linux,my fan works very well. . .
Is this technically explainable?
Thank you.

wiiznokes, 2025-02-01:
Are there any logs?

> Although this has nothing to do with this problem, I would like to ask a question. I have a dual system of windows and linux. My fan management seems to depend on windows?
just now i reboot several times in linux,and the fan kept working at full speed.
this is why I'm looking for a fan control software.
After I start windows once and reboot into linux,my fan works very well. . .
Is this technically explainable?
Thank you.

Although i never had this problem, i had some problem because of dualbooting, notably bluetooth, partition not mounted, wrong clock, so i'm not really surprised

yuzujr, 2025-02-01:
change behavior and switch of `Control` get this:
ERROR data::update] Can't set control to auto in set_invalid_controls_to_auto fn: Node cc was invalid