wiiznokes/fan-control #193: AMD GPU speed control errors

Router-Jockey, 2025-04-30:
#### Bug Description
AMD GPU fan reports speed correctly but unable to control.
Seems to be a permission issue. 
I did create the udev rule and control of chassis fans connected to the motherboard - driver/module nct6887 - do work. Only the GPU speed control does not.
Other applications work such as LACT. 

#### Expected Result
Control fan speed

#### Steps to Reproduce
Create a control targeting GPU and either a Graph or Flat.

#### Additional Information
LACT is able to manipulate this, however it requires Overclocking to be enabled which then allows for a non-zero RPM. I set this with LACT and can control the speed there without issue.
Fancontrol reports the fan RPM correctly but is unable to make changes. I stopped the LACT service and restarted Fancontrol, the problem persists. Perhaps this is an issue which is outside of the intended scope for Fancontrol, I'm not sure. 

#### Operating system
Kubuntu 25.04

#### fan-control version
fan-control 24.8.23

#### Logs

[ERROR data::update] can't update node: can't set value 244.79999999999998 to a pwm: sensors_set_value() failed: [9] Can't write
[ERROR data::update] can't update node: can't set value 244.79999999999998 to a pwm: sensors_set_value() failed: [9] Can't write
[ERROR data::update] can't update node: can't set value 244.79999999999998 to a pwm: sensors_set_value() failed: [9] Can't write
[ERROR data::update] can't update node: can't set value 244.79999999999998 to a pwm: sensors_set_value() failed: [9] Can't write
[ERROR data::update] can't update node: can't set value 247.35 to a pwm: sensors_set_value() failed: [9] Can't write
[ERROR data::update] can't update node: can't set value 247.35 to a pwm: sensors_set_value() failed: [9] Can't write
[ERROR data::update] can't update node: can't set value 247.35 to a pwm: sensors_set_value() failed: [9] Can't write
[ERROR data::update] can't update node: can't set value 247.35 to a pwm: sensors_set_value() failed: [9] Can't write
[ERROR data::update] can't update node: can't set value 244.79999999999998 to a pwm: sensors_set_value() failed: [9] Can't write
[ERROR data::update] can't update node: can't set value 244.79999999999998 to a pwm: sensors_set_value() failed: [9] Can't write
[ERROR data::update] can't update node: can't set value 244.79999999999998 to a pwm: sensors_set_value() failed: [9] Can't write
[ERROR data::update] can't update node: can't set value 244.79999999999998 to a pwm: sensors_set_value() failed: [9] Can't write
[ERROR data::update] can't update node: can't set value 244.79999999999998 to a pwm: sensors_set_value() failed: [9] Can't write
[ERROR data::update] can't update node: can't set value 244.79999999999998 to a pwm: sensors_set_value() failed: [9] Can't write
[ERROR data::update] can't update node: can't set value 244.79999999999998 to a pwm: sensors_set_value() failed: [9] Can't write
[ERROR data::update] can't update node: can't set value 244.79999999999998 to a pwm: sensors_set_value() failed: [9] Can't write
[ERROR data::update] can't update node: can't set value 244.79999999999998 to a pwm: sensors_set_value() failed: [9] Can't write
[ERROR data::update] can't update node: can't set value 244.79999999999998 to a pwm: sensors_set_value() failed: [9] Can't write
[ERROR data::update] can't update node: can't set value 247.35 to a pwm: sensors_set_value() failed: [9] Can't write
[ERROR data::update] can't update node: can't set value 247.35 to a pwm: sensors_set_value() failed: [9] Can't write
[ERROR data::update] can't update node: can't set value 247.35 to a pwm: sensors_set_value() failed: [9] Can't write
[ERROR data::update] can't update node: can't set value 247.35 to a pwm: sensors_set_value() failed: [9] Can't write
[ERROR data::update] can't update node: can't set value 247.35 to a pwm: sensors_set_value() failed: [9] Can't write

On linux, run `flatpak run io.github.wiiznokes.fan-control` in a terminal.

wiiznokes, 2025-04-30:
I'm not sure, maybe it's a flatpak limitation, maybe the udev rules are wrong

wiiznokes, 2025-04-30:
If lact run with admin permission, that could explain why it works for the app.
Also, did you change the udev rules file? If so, what did you write in it?

Router-Jockey, 2025-04-30:
OK yes LACT relies on a daemon which I had forgotten about, so yes it appears to be a permissions issue.

The udev rules I only used the example provided;

wget https://raw.githubusercontent.com/wiiznokes/fan-control/master/res/linux/60-fan-control.rules
sudo mv 60-fan-control.rules /etc/udev/rules.d/
sudo udevadm control --reload-rules && sudo udevadm trigger

Everything else works as expected.

wiiznokes, 2025-04-30:
We can leave this open, in case this is a udev rules problem